pinout-xyz/pico.pinout.xyz #6: [Feature Request]

HarvsG, 2022-12-02:
1. Support PICO W
2. label the LED with it's pin number - 32 for Pico W,  (25 for Pico)

lurch, 2022-12-03:
1. The user-accessible GPIO pins on the Pico W are the same as for Pico. See https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html
2. The pin number for the LED on Pico W isn't 32 - instead it's controllable via the WiFi chip. See https://github.com/raspberrypi/pico-examples/tree/master/pico_w/blink

Gadgetoid, 2023-02-23:
If I can get a good SVG for the front/back of the Pico W then it would be nice to add the option to switch between either-or. Even if that just hits different HTML pages.

I've raised #13 to ruminate about documenting extra pin functions, since I think these could be useful to surface outside of the datasheet(s). Granted I am keen to keep things *clean.*

Gadgetoid, 2023-04-25:
Done: https://picow.pinout.xyz/

And done: https://pico.pinout.xyz/

Check "Advanced" in both cases to see the additional tables. I realise that the user LED isn't all that advanced on Pico, but I didn't want things to get too visually cluttered.

closed 2023-04-25 by Gadgetoid

lurch, 2023-04-25:
Doesn't seem 100% reproducible, but sometimes goes a bit funny when switching between the two sites? (Google Chrome on Linux)

[image]

Gadgetoid, 2023-04-25:
That's the old CSS still lurking about in the cache, a bit of shift + F5 action usually clears it up. If it persists I might need to drop a cache-buster in the CSS URL for good measure :laughing: 

There are only two really hard problems in programming:

1. Naming things
2. Cache invalidation
3. Off by one errors

lurch, 2023-04-25:
3. Forgetting to start at index 0