pinout-xyz/pico.pinout.xyz #8: Mislabelled pin function

lurch, 2023-02-03:
When you click on the "Advanced" box on https://pico.pinout.xyz/ it shows GP18 as "UART1 RTS", but page 18 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf says GP18 is "UART0 CTS".

lurch, 2023-02-03:
Huh, I've just noticed that the mouse-hover text for those pins actually seems correct :laughing:

Gadgetoid, 2023-02-03:
What's the deal with the tooltip on the pin above it though, aaaaaagh!

Maybe I should be generating this HTML from a dataset.

lurch, 2023-02-07:
Looks like the hover-text for GP27's UART mode is wrong as well? The page says "UART1 RTS" (which agrees with https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf ) but the hover-text says "UART 0: Clear to send".

Gadgetoid, 2023-04-25:
Finally fixed this last little bug. Hopefully that's all of them!

closed 2023-04-25 by Gadgetoid