pinout-xyz/pico.pinout.xyz #13: Document GPIO23, 24 and 25

Gadgetoid, 2023-02-23:
Something of a dupe of #6 except it's a different approach.

Might be handy to have an addendum at the bottom of the page, a small table that covers these additional pins and their functions. This could include the SWD pins for #7 

* GPIO23 is connected to the PS pin of the onboard RT6150B-33GQW, if it's LOW (default) it's in PFM mode for best efficiency, if it's HIGH it's in PWM  mode for improved ripple but worse efficiency at light loads. See: https://datasheets.raspberrypi.com/pico/pico-datasheet.pdf#pico-schematic-diagram

* GPIO24 is the VBUS sense pin, if VBUS (either via USB, or the VBUS pin) is present then it's HIGH, otherwise LOW.

* GPIO25 on Pico is the USER LED, and on Pico W it's the wireless SPI chip-select. See: https://datasheets.raspberrypi.com/picow/pico-w-datasheet.pdf

lurch, 2023-02-23:
Yeah, see also section 2.1 of https://datasheets.raspberrypi.com/picow/pico-w-datasheet.pdf

samveen, 2023-04-10:
Slightly related, but not much, maybe separate out the [PiCow](https://www.raspberrypi.com/app/uploads/2022/07/PicoW-800x533.jpg) into it's own correl at picow.pinout.xyz ??

lurch, 2023-04-11:
> maybe separate out the PiCow

Isn't that what #6 is about? Just out of curiosity, what specific difference between the Pico and Pico W do you think needs clarifying?

samveen, 2023-04-17:
@lurch The board layout of the 2 is different (not the pins, but the components), so it might be a tad disconcerting for newbie users (who are probably the primary target audience of pinout), and thus leading to extra time wasted in figuring things like the location of the debug headers being a cosmetic change.

lurch, 2023-04-18:
I'm surprised to hear that it took you a whole hour - did you look at https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html ?

samveen, 2023-04-18:
No, I didn't . I do wish google had given me this result first. I was muddling through comparing multiple pinout diagrams on the datasheets subsection of the raspi website.

Gadgetoid, 2023-04-20:
I’m not opposed to a PicoW version of the site for clarity. Now that I’ve moved the JS and CSS to separate files it’s rather less onerous to maintain multiple variations. I really should distill the pin details to some database format and generate the HTML from that, though.

I’m not sure how easy it would be to convince Google to serve the right version, but we can -  at least - interlink the two. They might also need to be subfolders rather than subdomains to more easily support installation as a PWA and preserve the ability to view each respective board offline. Though I guess two separate PWA’s for separate boards (ie: two icons on your desktop or phone) is a nice separation of concerns?

samveen, 2023-04-21:
_I should probably apologize for my little side comment about a variant sub-site for the Pico W hijacking this issue_  .

Given that the search term combination for 'pinout+pico' already lead to the correct location with Google, I'm expecting a new subdomain like picow.pinout.xyz will be among the top results for 'pinout+pico+w' very quickly, if interlink from the pico page.

Gadgetoid, 2023-04-24:
I've cleared the first hurdle- getting front/back SVG files for the Pico W which are here:

https://raw.githubusercontent.com/pinout-xyz/pico.pinout.xyz/pico-w-prototype/raspberry-pi-picow.svg

and here:

https://raw.githubusercontent.com/pinout-xyz/pico.pinout.xyz/pico-w-prototype/raspberry-pi-picow-underside.svg

Gadgetoid, 2023-04-24:
And - if we're adding a whole new table - I don't think it would be unreasonable to link to the respective documentation and datasheets for both boards? ... at least *somewhere*

samveen, 2023-04-25:
Need a link on  the Pico page  ([View the Pico W version of this pinout.](https://picow.pinout.xyz/)) similar to the link on the PicoW page ([View the Pico version of this pinout.](https://pico.pinout.xyz/))

Gadgetoid, 2023-04-25:
Okay, I've rolled out some extra labels for the debug pins and a little table documenting GP23, GP24 and GP25 as noted above, plus GP29 and ADC4 which I had missed.

[image]

Gadgetoid, 2023-04-25:
And on the Pico W side of things:

[image]

samveen, 2023-04-26:
Some formatting issues crop up in Firefox 112.0.1 ((x)Ubuntu 20.04) as below:

[image]

lurch, 2023-04-26:
https://github.com/pinout-xyz/pico.pinout.xyz/issues/6#issuecomment-1522436645

:wink:

lurch, 2026-08-16:
> plus finding umpteen other instances of breakage and drift

Including this, which I spotted yesterday: https://picodocs.pinout.xyz/
It only covers pico-sdk 1.5.0 when pico-sdk 2.3.0 is the latest version, and it _probably_ isn't necessary anyway in light of https://www.raspberrypi.com/documentation/pico-sdk/ ?

Gadgetoid, 2026-08-16:
I ought to retire that, or perhaps regenerate. I created it because RPi's own documentation format is an infinite scrolling page, making it a very long game of "guess if you're still reading docs about the thing you clicked on."

Couple that with the mostly featureless white on black text and it's a dense, unparseable mess with no visual structure or landmarks and I find it thoroughly and completely inscrutable.

I usually just read the Pico SDK sourcecode if I need to know anything so I'd forgotten about picodocs 😆 

Unless you're some kind of text processing savant, then this:

[image]

Is *vastly* superior to this:

[image]

So much so I genuinely wonder if I'm seeing a broken version of the website!?

Gadgetoid, 2026-08-16:
muwahaha https://picodocs.pinout.xyz/versions.html

lurch, 2026-08-17:
If you have a complaint about the formatting of the pico-sdk docs, please create an issue at https://github.com/raspberrypi/documentation/issues 😉   Or I guess you could just add additional comments to https://github.com/raspberrypi/documentation/issues/4156

lurch, 2026-08-17:
> muwahaha https://picodocs.pinout.xyz/versions.html

Ah! I've just noticed that this is the "RP2040-only" version of the pico-sdk docs, so is missing all the RP2350-specific functionality, e.g. compare https://picodocs.pinout.xyz/group__multicore__doorbell.html and https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#group_multicore_doorbell
(But note also that there are some functions which are RP2040-specific, e.g. https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_hardware_rtc )

Have a look at Appendix B of https://rptl.io/pico-c-sdk for instructions about building the "combined docs".

Gadgetoid, 2026-08-18:
At the risk of continuing this slightly off-topic debacle and consequential nerd-snipe- https://picodocs.pinout.xyz

I added a drop-down to pick between targets (so chip specific, rather than combined), carefully refined the whole build process (down from about 1+ hours to 30 odd seconds), and deployed a bunch of monkey patches to fix the search box and supplant the trademarked logo.