pinout-xyz/pico.pinout.xyz #14: Missing advanced functions on GP26, 27, & 28

ZodiusInfuser, 2023-03-03:
These pins have SPI functionality that the advanced view does not show.

[image]

Is this an easy enough thing for me to edit and raise a PR for? Looks like they would need a 5th column on the UI, so perhaps beyond me web skills?

Gadgetoid, 2023-03-03:
Whew, that's a rather awkward problem. I might have to eschew the original pinout diagram layout in order to better incorporate the ADC functions.

I guess `Ground / AGround` is rather redundant, and `ADC_VREF1` being moved one spot over is also slightly weird.

What if ADCn was combined with `GPn` in the first label, recoloured to ADC colour and SPI got its rightful slot.

Like this:

[image]

ZodiusInfuser, 2023-03-03:
That compacting seems okay to me, but then I'm not the primary audience of the non-advanced mode.

Gadgetoid, 2023-03-03:
I don't like the compacting much, but I think it's a necessary evil to avoid occluding pin functions.

All the I2C, SPI and UART RX/TX that's currently not shown by default can be added and make the default view a little more visually balanced :laughing:

lurch, 2023-03-03:
Works for me :wink:

Gadgetoid, 2023-03-03:
Proposed default view:

[image]

The "ADC" toggle becomes redundant... well I say "becomes", it was never all that useful.

Still need to address #13

closed 2023-03-03 by Gadgetoid

Gadgetoid, 2023-03-03:
Ugh, darn CSS changes are being stubborn about re-loading. Might need a cache-buster. Almost as frustrating to refresh as 7-colour e-Ink :skull:

PR #15 (2023-03-03): Chris.
Merge ADCn pin labels with GPn labels. Add missing SPIn functions. Make some "Advanced" labels default.

(Fixes #14)