pinout-xyz/pico.pinout.xyz #18: pinout is wrong

bablokb, 2023-11-08:
Comparing your pinout with the official pico-datasheet, your pinout labels some pins wrong:

- GPIO20/21: pico-datasheet does not list them as UART
- GPIO20: not a SPI-RX pin 
- GPIO26-28: no SPI pins

Not sure if this is complete or not...

lurch, 2023-11-08:
I just checked, and the diagram at https://pico.pinout.xyz/ **is** in fact correct - if you look at the table on page 13 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf you can see that the "errors" you list above _are_ actually correct.

The pinout diagram in https://datasheets.raspberrypi.com/pico/pico-datasheet.pdf (and at https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#pinout-and-design-files ) chooses to omit the `UART1 TX` function from GP20 because GP23 isn't user-accessible on the Pico board, and so the "full set" of UART pins (TX, RX, CTS, RTS) on GP20-GP23 isn't user-accessible on the Pico. And similar logic explains the SPI pins you mention too.

lurch, 2023-11-08:
It's not _quite_ the same thing that you're asking, but see also https://github.com/raspberrypi/pico-feedback/issues/193

bablokb, 2023-11-08:
This is a real pity and confuses users. The official datasheet does not list CTS/RTS at all, so omitting these pins does not make sense to me.

So maybe adding a note to  https://pico.pinout.xyz/  might help? E.g. "This pinout also shows available functions not listed in the Pico Datasheet")?

Gadgetoid, 2023-11-08:
Wait, we're more correct than the official documentation? NEVER! :laughing: 

A little asterisk on unlisted functions and a footnote might work. Though I'm apprehensive to add too much clutter.

lurch, 2023-11-08:
One person's "more correct" is another person's "overly verbose" :wink: :shrug: