pinout-xyz/pico.pinout.xyz #19: Pico 2

cfpwastaken, 2024-08-09:
The new Pico 2 seems to have one more Test Pad on the bottom. It should be documented.
In addition, you can also redirect USB DP/DM to any GPIO. Maybe show that in advanced mode too?

Gadgetoid, 2024-08-09:
Made the repo and started some initial stuff here- https://github.com/pinout-xyz/pico2.pinout.xyz

cfpwastaken, 2024-08-09:
The new 7th test pad is labelled as "1V1 - do not use" in the datasheet. I think it should still be documented. However, i cant find the place where the other TestPads are located, was this never a thing?

cfpwastaken, 2024-08-09:
seems like testpads being listed actually never was a thing. Should I PR it on all repos?

Gadgetoid, 2024-08-09:
I think it’s worth doing- gated under Advanced.

AIUI the 1v1 pad can be used to inject MOAR VOLTS for ambitious overclocks. The core voltage for the Pico 2 can be cranked distressingly high.

cfpwastaken, 2024-08-09:
aha. I guess ill just add it to the advanced table

cfpwastaken, 2024-08-09:
=> #20

cfpwastaken, 2024-08-09:
and for the pico2 repo: https://github.com/pinout-xyz/pico2.pinout.xyz/pull/3
cant fork the picow repo since it says i already have it. Probably because its a fork of the pico repo which i do have. Need to wait until i can delete the pico one.

Gadgetoid, 2024-08-10:
> cant fork the picow repo since it says i already have it.

Curses, I should have hard-forked that. But thank you!

cfpwastaken, 2024-08-10:
ok. here is the picow pr: https://github.com/pinout-xyz/picow.pinout.xyz/pull/1

cfpwastaken, 2024-08-10:
i also noticed that the Wifi GPIO pins are actually in a different table to group it away from the others. shouldve done it there too. Dont think this is worth tho right?

closed 2025-03-11 by cfpwastaken